xela144/CANaconda #72: Used a fixed-width representation for all data

Susurrus, 2015-01-07:
This makes comparing values much easier. Basically the smallest decimal point should be selected based on the scaling value for a given field, with an equivalent calculation done for integer values, and a fixed-width representation should be chosen. If the values are also right-aligned, then it'll be much easier to compare subsequent values, which becomes difficult when values jump between things like "12.4" and "12.50000000000001" or other values which have a small amount of precision and then a high amount of precision or vice-versa.

For example, for a signed 8-bit field that stores units with a scaling of "0.1", the possible output range is from -12.8 to 12.7, so it should always be output as 5 characters (1 for sign, 2 for integer portion, 1 for the decimal, and 1 for the fractional portion. Smaller values should be displayed without 0s prefixed, so 0 would show as "  0.0".

xela144, 2015-01-10:
Try it now, and see if it is good enough. If it still needs improvement, I will further investigate string formatting options.

xela144, 2015-01-19:
Can you let me know if the current format works for you.

Susurrus, 2015-01-19:
I'll take a look at this Tuesday, when I'm back in lab and have the hardware.

Susurrus, 2015-01-26:
Fixed as of 505c6ef

closed 2015-01-26 by Susurrus