xela144/CANaconda #77: View and update main README.md

Susurrus, 2015-01-17:
Run through all of the docs in the README and verify that they are correct and update. If detailed installation instructions are provided, test them in a VM and indicate which version of the OS they worked for.

Additionally the following should be added/changed in the README:
 * The Lawicel CANusb drivers aren't relevant, so remove them as a requirement. Also the serial number for the device isn't relevant either, so remove it.
 * System requirements should list the supported Python versions. I think 3.3+ would work, correct?
 * Add documentation for all stand-alone tools. This means pipePlotter should have its own section. And any other tool that can be run by itself on the command line (I don't think there are any, but there sure are a lot of Python files, so I don't fully know).
 * The `metadata/` folder isn't talked about at all in the README. You should move all XML format stuff to that folder's README and just describe that XML files exist there to be used, but custom ones can be used as well. Be sure to reference that README file from this one.

xela144, 2015-01-19:
Have a look... and feel free to re-open.

closed 2015-01-19 by xela144

xela144, 2015-01-19:
[link to metadata folder](https://github.com/xela144/CANaconda/tree/master/metadata)