xela144/CANaconda #81: Nothing is output when metadata file fails parsing

Susurrus, 2015-01-25:
It'd be nice to have some output regarding what went wrong to help debugging the XML files. The SeaSlug.xml file no longer parses correctly with the latest pull, but I'm unsure what is causing it. Debugging this is painful.

xela144, 2015-01-25:
Is it the two new IMU messages? Did you fix it by specifying endianness?

If this is the case, then the error should have been handled here:
https://github.com/xela144/CANaconda/blob/1326d415d947e8b3ff982036611fbf1c3fbb5b7f/messageInfo.py#L151-L154

The first 'if' is for syntax checking.
The second 'if' is for inferring the endian-ness.

Sounds like this wasn't working?

Susurrus, 2015-01-26:
Should be fixed in: 240f20e. Probably worth checking that all other error messages can be triggered and don't end up being caught and made useless.

closed 2015-01-26 by Susurrus