xela144/CANaconda #86: can you add support for Beagle bone black

wulfmans, 2015-04-26:
The Beagle bone black has a built in CAN bus interface. Your program is really cool and i want to use it on my BBB as a Xanbus system monitor but i am unable to code in python and have no idea how to tell your program to use CAN0 as the input device rather than the serial/usb port.
I would have emailed the creator of the program but was unable to fine your address  thanks a bunch
my git username is wulfmans

JEB12345, 2015-04-27:
I have not implemented CANaconda on the BBB, though I use CAN on the BBB extensively. In order to setup the CAN bus module, follow the instructions posted here [here] (http://www.embedded-things.com/bbb/enable-canbus-on-the-beaglebone-black/).

Make sure to install the terminal based CAN utilities mentioned at the end of the article. This is the basic way you will monitor and interact with the CAN bus.

wulfmans, 2015-04-27:
no need to use the instructions you supplied
the can cape i have from logic supply does all the heavy lifting

once installed all i have to do is is use the candump can0 command to
see the data

can0  09F01D01   [4]  33 01 9E 00
  can0  19F0C501   [8]  60 15 03 03 0E 60 00 00
  can0  19F0C501   [8]  61 C2 CF FF FF 2F 01 00
  can0  19F0C501   [8]  62 00 FF FF FF FF FC 00
  can0  19F0C501   [8]  63 00 FF FF FF FF FF FF
  can0  18EEFF00   [8]  25 72 FF FF 00 80 3C 80
  can0  0DF80900   [8]  03 21 9B 3C 55 00 00 F9
  can0  09F01D01   [4]  33 01 94 00
  can0  19F0BF02   [4]  03 20 03 00
  can0  19F00F01   [6]  03 04 03 03 02 00
  can0  19F0C402   [8]  20 24 03 03 2C 60 00 00
  can0  09F01D01   [4]  33 01 9B 00
  can0  19F0C402   [8]  21 00 00 00 00 00 00 00
  can0  19F0C402   [8]  22 00 FF FF FF FF 00 E4
  can0  19F0C402   [8]  23 70 3D FF FF FF FF FF
  can0  19F0C402   [8]  24 FF FF FF FF FF FF FF
  can0  19F0C402   [8]  25 00 FF FF FF FF FF FF
  can0  19F00F02   [6]  03 04 03 03 02 00
  can0  18EEFF01   [8]  4A B2 EE FF 00 81 3C 80
  can0  09F01D01   [4]  33 01 94 00
  can0  19F00E02   [8]  20 11 03 03 00 00 00 00
  can0  19F00E02   [8]  21 00 00 00 00 64 01 1D
  can0  19F00E02   [8]  22 03 02 C5 FF FF FF FF

cansniffer cant seem to parse that data stream

what command line command can i use with your program to deal with that data
./canpython.py --nogui /dev/ttyUSB0 -m xmltest.xml
--filter='WSO100{airspeed},WSO200{wind_dir=2,velocity}' --csv --time --zero

obviously will not work as i don't use the serial to can adapter

i tried
./CANaconda.py --nogui -p can0 -m metadata/Xanbus.xml  --csv --time --zero

can0 is not a /dev/
its like eth0

with no success

I am a hardware engineer and sometimes the command lines for commands
just do not make sense to me

xela144, 2015-04-27:
Hi there. Thanks for your interest! Yes, the program looks for a serial connection over USB. Since this is coded in Python, we are limited to what Python modules can see. Pyserial is what we're using to connect over USB. 

Can you provide me with more info on can0, so that I can look into it? If it is like eth0, then it looks like we can use the 'socket' module, and create a socket connection. Then we should be able to add support for the BBB.

One note on Xanbus: They use NMEA2000 fast packets. A lot. Currently this project does not support fast packets, although we are planning on adding this. See issue #49

wulfmans, 2015-04-27:
yes its exactly like a socket connection

there are can tools for linux and they have command line interface with
tools like candump cansniff and cansend

you use them like "candump can0"

can0  09F01D01   [4]  33 01 BF 00
  can0  19F00F02   [6]  03 04 03 03 02 00
  can0  09F01D01   [4]  33 01 B5 00
  can0  19F00F00   [6]  03 04 03 03 03 40
  can0  18EEFF02   [8]  11 E5 EE FF 08 81 3C 80
  can0  09F01D01   [4]  33 01 BB 00
  can0  19F0BF01   [4]  03 20 03 00
  can0  09F01D01   [4]  33 01 BB 00
  can0  19F0C401   [8]  40 24 03 03 52 6C 00 00
  can0  19F0C401   [8]  41 F2 D1 FF FF 46 01 00
  can0  19F0C401   [8]  42 00 FF FF FF FF 00 48
  can0  19F0C401   [8]  43 71 64 FF FF FF FF FF
  can0  19F0C401   [8]  44 FF FF FF FF FF FF FF
  can0  19F0C401   [8]  45 00 FF FF FF FF FF FF
  can0  19F0BF02   [4]  03 20 03 00
  can0  09F01D01   [4]  33 01 BF 00
  can0  19F00F01   [6]  03 04 03 03 02 00
  can0  19F0C402   [8]  40 24 03 03 34 6C 00 00
  can0  19F0C402   [8]  41 00 00 00 00 00 00 00
  can0  19F0C402   [8]  42 00 FF FF FF FF 00 48
  can0  19F0C402   [8]  43 71 64 FF FF FF FF FF
  can0  19F0C402   [8]  44 FF FF FF FF FF FF FF
  can0  19F0C402   [8]  45 00 FF FF FF FF FF FF

this is raw can data

sometimes i wish i did programming a bit more but i have been into
hardware for 40 years and i will prolly stay a hardware engineer

wulfmans, 2015-05-10:
Well here is some information for upgrading this to run on the beagle bone.
http://en.wikipedia.org/wiki/SocketCAN
Python added support for SocketCan in version 3.3.[1] An open source library python-can provides socketcan support for Python 2.
https://bitbucket.org/hardbyte/python-can

I will poke around to see how to change out the code to use this.

Also how were you able to find that one XML file for the xanbus ?
I have now decoded all the commands and have the payload information to decode messages
can1  0DF80900   [8]  03 88 E2 4F 55 5C FE F8
0x0DF80900 PGN 129033 Date and Time Status
first byte of payload i am unsure of next 4 bytes in reverse 554fe288 is the unix hex time
GMT: Sun, 10 May 2015 22:58:16 GMT
Your time zone: Sunday, May 10, 2015 3:58:16 PM
Decimal timestamp/epoch: 1431298696

i think the next 2 bytes are the GMT offset and i am unsure of the last byte.

i can provide the data to write the Xanbus XML files as i cant quite wrap my head around creating that.
I believe that you can read the canbus with the beagle fast enough for nema fast packet by
setting up the can interface like this.
ip link set can1 type can bitrate 250000 triple-sampling on
I have logged 1000s of frames with 0 dropped frames.

I noticed that you only use QT5 for the gui
My BBB only has QT4 installed and will need to wait for the upcoming Jessie release for the
upgraded software. I upgraded it and had to back out the QT5 as it broke things.

I am always glad to help on this project as its really cool

Susurrus, 2015-05-11:
Then it sounds like generically we could support SocketCAN with this software and get BBB support "for free" along with supporting a wide variety of CAN interfaces on Linux. That seems like a win.

An example of doing this with Python 3.3 is: https://libbits.wordpress.com/2012/05/22/socketcan-support-in-python/

Note that we will never support anything below Python3.3 and Qt5, it's not worth our development time and many systems already support these versions. That being said, you don't need the GUI to run CANaconda, see the `--nogui` option to stream CAN output to the terminal.

wulfmans, 2015-05-11:
how do you add to the XML file ?
i can start with the small commands and work my way up to the larger ones

Susurrus, 2015-05-11:
See the XML files in the `/metadata` folder. The README also describes their format. They can be located anywhere, but we keep ours in that folder.

xela144, 2015-05-12:
@wulfmans I don't have a BBB, although I would like to get one to bring up support for that interface. For the time being, however, I won't be able to modify the project to make it work for you. 

You said that you are a hardware guy and so you probably won't write any code yourself. But if you can find someone who is willing to help, I can point to where the changes/additions need to be made. 

Currently we use a serial connection and the Lawicel CAN to USB device, and the code for this is in [canport.py](https://github.com/xela144/CANaconda/blob/master/canport.py). So that is where we would start. Rather than using serial and Lawicel, we would use SocketCAN and BBB. 

Have a look at [this diagram](https://github.com/xela144/CANaconda/blob/master/app%20architecture.png). I just included it in the repo today. This will give you a rough idea of how the application works.

wulfmans, 2015-05-12:
I can send you one if you would like
i have an extra one i can load up BBB linux on it if i can find an extra
micro sd card and send it to you for development
if not you can grab a 8 or 16 gig sd card for it and get the latest
release of software.
I got a few python books and will bone up on coding and can also take a
look at it
right now i am working on the XML files for Xanbus. some of the data is
a guess
as to the length of the data fields. but i will make sure the data adds
up to the total per command
being semi retired is nice for available time.
The more the Xanbus protocall is out there the more hacking people can
do on the xanbus
I love open source software. 30 years ago a fellow engineer said to me
charge for your hardware and make software free
its been working well since then.

wulfmans, 2015-05-12:
https://github.com/xela144/CANaconda/blob/master/app%20architecture.png
gives me a 404

would a candump data file help you ?
  can1  09F01D01   [4]  33 01 BB 00
  can1  19F0C502   [8]  00 15 03 03 00 5F 00 00
  can1  19F0C502   [8]  01 00 00 00 00 00 00 00
  can1  19F0C502   [8]  02 00 FF FF FF FF FC 00
  can1  19F0C502   [8]  03 00 FF FF FF FF FF FF
  can1  19F00301   [8]  E0 44 03 13 01 01 00 00
  can1  19F00301   [8]  E1 00 00 00 00 00 00 00
  can1  19F00301   [8]  E2 00 B8 0B 00 00 00 00
  can1  19F00301   [8]  E3 00 00 00 00 02 01 00
  can1  19F00301   [8]  E4 00 00 00 00 00 00 00
  can1  19F00301   [8]  E5 00 00 B8 0B 00 00 00
  can1  19F00301   [8]  E6 00 00 00 00 00 03 01
  can1  19F00301   [8]  E7 00 00 00 00 00 00 00
  can1  19F00301   [8]  E8 00 00 00 B8 0B 00 00
  can1  19F00301   [8]  E9 00 00 00 00 00 00 FF
  can1  09F01D01   [4]  33 01 C8 00
  can1  19F00F01   [6]  03 04 03 03 02 00
i can send a few megs of a dump off the live data stream

also i was reading about the XML format
*"size* - The size of the message payload, in /bytes/. Valid values: [0,
8]."
the xanbus data has 4 byte payloads like 09F01D01 and like 19F00301
above, a sequence of 8 byte payloads
with the first byte being a device sequence of 3 bits that changes every
message of that type sent
and 5 bits that is the message count
can i change that size in the descriptor to 4 and 6 to accommodate  the
other sizes or is that hard coded somewhere

the date and time data  "can1  0DF80900   [8]  03 22 23 51 55 5C FE F9"
first byte i am unsure of  the next 4 bytes is the unix timestamp in hex
but in REVERSE byte order
does canaconda  have a way to reverse the payload data if using xanbus ?
the next 2 bytes are the GMT offset. I know the offset is there but cant
figure out what the numbers should be
most offsets are in seconds but playing with the data a bit did not make
any sense to me surely not in seconds ?
maybe a hard coded offset
i can dump the data and try all offsets for a table
                                     hex         dec
00 00    0:00 gmt    0x0000    0000
C4 FF   -1:00 gmt    0xFFC4    4036     
88 FF   -2:00 gmt    0xFF88    3976
4C FF   -3:00 gmt    0xFF4C    3916
2E FF   -3:30 gmt    0xFF2E    3386
10 FF   -4:00 gmt    0xFF10    3856
F2 FE   -4:30 gmt    0xFEF2    3826
D4 FE   -5:00 gmt    0xFED4    3796
98 FE   -6:00 gmt    0xFE98    3736
5C FE   -7:00 gmt    0xFE5C    3676

i know this as i dumped the data while i was changing the GMT offset on
the system control panel

Last time i did any programming was C code on a Univac 1100 in the early
80s   kinda dates me i know
python does not seem that impossible to learn with a little effort and
sure easier than C coding

your hardware files on github. what schematic package was used to create
them. eagle? i use Orcad and Allegro
maybe i can redraw them in that format for an alternate package to add
for your project.

I was thinking about my offer of a BBB, i dont mind sending one but it
will do you no good
unless you have a can cape and a live data source which i have.
there is a serial port on the BBB and if you have a data source and a
serial adapter you can use it that way
and get the can interface working on a Vcan.
soon they will be using a new kernel and i will get it and upgrade
everything i can to see if i can get the gui working
on the new image.

wulfmans, 2015-05-13:
IS there a way for CANaconda to read the can data from a file ?
that would be a great way to troubleshoot the XML files from a data dump
if you dont have a can bus source
hooked up to your computer.
its going to take me another week to decode the xanbus pgn data but i
have a lot done already

Susurrus, 2015-05-13:
I thought you could pipe a file directly into the program when run 
without the GUI like:
`cat MY_FILE | canaconda --nogui --metadata=metadata/xanbus.xml`

The README and the --help option should give you more details.

wulfmans, 2015-05-13:
ERROR in data backend: Please specify a port (using --port option)

On 5/13/2015 8:43 AM, Susurrus wrote:
> cat MY_FILE | canaconda --nogui --metadata=metadata/xanbus.xml

Susurrus, 2015-05-13:
Like I said, I wasn't sure. Read the documentation to confirm. Otherwise I'll let @xela144 chime in when he can.

xela144, 2015-05-13:
@Susurrus Yeah, the program will look for a serial port when you run it from the command line. The pipe was used for saving .csv files, or to plot data using the matplotlib script.

@wulfmans We never had to parse a text file in development, so this was never a feature. What you are trying to do is change the parsing algorithm. This is because the candump format is different from Lawicel. Lawicel just gives ascii strings: [ID][LENGTH][PAYLOAD]. The parsing breaks up the payload by offset. The offset is given in the user xml file.

So it sounds like there are two issues you are having. The first is to understand the bit offset within the payload. This part is on you, as the user, because those offsets are what you put in the xml file. Without the xml file, the program won't know what to do with the payload. Endianness of that data should not be an issue, since it is specified by the protocol (Xanbus is NMEA2000, right?), and the parser knows about this.

The other issue is the format. candump is not sending a raw ASCII representation of the message stream. It is already doing some parsing for you. So there are ways to solve this. 

1. Modify candump so that it just returns raw ASCII strings formatted as [ID][LENGTH][PAYLOAD]. Then package this up in the project and release it.
2. Modify the canaconda parser so that it knows about the candump format.

I won't be able to do this until I get my hands on a BBB and see for myself what is going on. I should be able to do this in the next week or so. Also, I'm not entirely convinced that we will use candump in this project, because it seems to be for listening only. We want to be able to transmit messages also. In that light, I'd prefer to use something similar to what Lawicel does, because it makes transmission easier. We just package up a message in raw ASCII, and let the device handle the CAN stuff.

If you are interested in knowing the secret sauce for parsing Lawicel raw ASCII messages is [here](https://github.com/xela144/CANaconda/blob/master/canport.py), look for the regex parser in pyserialInit function. As you can tell (by the name of that function), the parser and the serial connection are coupled. So there will have to be some refactoring to get the BBB supported. This is a good thing, since by supporting two interfaces, adding a third or fourth will be superficial. Still, lot's of work to be done. :hammer: 

PGN parsing is handled by the code also. See [Nmea200.py](https://github.com/xela144/CANaconda/blob/master/Nmea2000.py)

The schematic you asked about is done in Eagle. We haven't sent any boards out yet, and that's why there is no layout yet. For that hardware, we'd also need some C code to be written on the PIC.

xela144, 2015-05-13:
@wulfmans That png file should be there. Try this link: https://github.com/xela144/CANaconda/blob/master/application architecture.png

wulfmans, 2015-05-13:
well there is cansend candump and cansniff in can-utils. but i think the
thing you want is to use the kernel
hooks called socketcan if i read correct the python3.4 already has
support in it for socketcan

I can send an original beagle bone for testing as long as i get it back.
its cheap postage as its a small board
just email me an address

you will need a cancape from logic supply or one you can get on ebay
from china
and of course a can data stream

here is a sample xml for the date and time

    <messageInfo name = "Date and Time Status" pgn = "129033" size = "8"
protocol = "nmea2000" type = "fast-packet">
        <desc>Date and Time.</desc>
        <field
        name = "Device sequence"
        type = "enum"
        offset = "0"
        length = "2"
        signed = "no"
        />
        <field
        name = "Message count"
        type = "int"
        offset = "2"
        length = "6"
        signed = "no"
        />
        <field
        name = "Unix hex timestamp"
        type = "int"
        offset = "8"
        length = "32"
        />
        <field
        name = "GMT Offset"
        type = "int"
        offset = "40"
        length = "16"
        />
        <field
        name = "Unknown"
        type = "int"
        offset = "56"
        length = "8"
        />
    </messageInfo>

can1  0DF80900   [8]  03 20 FF 48 55 00 00 FB       "PGN_0x1F809 - Date
and Time Status"        *SYSTEM CONTROL PANEL*

the time is the 2nd 3rd 4th and 5th bytes BIG endian   so  the time is
5548ff20 unix time
the 6th and 7th are the GMT offset and in this packet its 0 as i set it
to GMT +0

i am working on the rest of the PGN codes i see on my network and have a
secret weapon for the decode
but it would be awesome to use the candump output as an input file for
CANaconda
maybe i can change a copy of candump to output the data as you described

xela144, 2015-05-14:
Good. Sounds like things are moving along. Let us know how it goes.

One thing about your metadata I noticed is that you are specifying type as fast-packets. That won't work, because we are talking about data types here. Supported types are 'int', 'bitfield', 'boolean' or 'enum'. In addition, fast-packets are not supported, yet. The good news is that your messages all seem to use regular CAN frames. 

About the BBB, we have obtained one here. No CAN cape, as it uses a custom board. Just waiting on a pinout from @JEB12345 (Not urgent.)

xela144, 2015-05-14:
Also, I think you are correct about using SocketCAN, as it is part of the Python socket module. I think we should support SocketCAN as a comm protocol, and then support BBB + CAN cape as a subset of that.

Susurrus, 2015-05-14:
We shouldn't need to support the BBB any differently than any other system with SocketCAN configured. It ends up being similar to a serial port, the user just need to specify the device. I'd suggest renaming the `--port` argument to `--device` so that it can be used to specify serial ports or SocketCAN sockets. Ideally we could also auto-detect which it is.

wulfmans, 2015-05-14:
I used the xanbus.xml file in your  metadata folder as a template
fast packets was enabled in there and i left it that way
i can change it to whatever works

what does your program do with the first byte of the payload ?
the first 3 bits increment every time a source sends it and the next 5
bites increment for every 8 byte message
sent in a PGN
i am unsure if i need to specify that in the XML file

Susurrus, 2015-05-14:
That's an NMEA2000 fast packet, so special handling of that isn't done 
yet in CANaconda.

xela144, 2015-05-14:
@wulfmans I see. It's in the repo like that. For now, you can leave that out, since it looks like you're not using fast packets.

Susurrus, 2015-05-14:
Also, to support fast packets, I have implemented C code that does it here: https://github.com/Susurrus/Autoboat/blob/master/Code/Libs/C/Nmea2000.c#L83, in the `Nmea2000FastPacketExtract()` function.

A straight port of this to Python, and then detecting FastPacket types in the XML file, so you know which messages to feed that decoding function, and we'd have support for Fast Packets. Shouldn't be that hard.

wulfmans, 2015-05-14:
it seems that xanbus uses fastpacket and single frame packets

i will set up the xml files with both or at least the best i can

i just got my xantrex mppd charge controller so more packets to decode

i also will be getting today a combox which decodes everything and gives
a nice web interface
I believe it uses a freescale processor running an imbedded linux of
some type.
i will take pictures and see if i can reverse engineer the device a bit
as well.
every little bit of info is good.

wulfmans, 2015-05-14:
an excerpt from     http://www.can-cia.org/index.php?id=490

NMEA 2000 allows the use of Fast-packet transmission for up to 256-byte
data with its own identity. This method allows sequential frames to be
sent; a single data byte in the first frame is used to specify the size
of the entire data and each additional frame uses one data byte as a
frame counter.  

the xanbus does not seem to follow nema 2000 fastpacket as the first
byte of the series of packets does not seem to have the byte count in it

  can1  19F0C401   [8]  00 24 03 03 4A 6A 00 00
  can1  19F0C401   [8]  01 66 D1 FF FF 44 01 00
  can1  19F0C401   [8]  02 00 FF FF FF FF 00 74
  can1  19F0C401   [8]  03 72 64 FF FF FF FF FF
  can1  19F0C401   [8]  04 FF FF FF FF FF FF FF
  can1  19F0C401   [8]  05 00 FF FF FF FF FF FF

this   0x1F0C4 - Battery Status just uses the last 5 bits of each
packet  as a frame counter

I think that on the xanbus they know what to expect since its a closed
system
the next 0x1F0C4 - Battery Status set of packets increments the first 3
bits seeming to indicate the new data
from the same device

  can1  19F0C401   [8]  20 24 03 03 30 61 00 00
  can1  19F0C401   [8]  21 06 C3 FF FF 84 01 00
  can1  19F0C401   [8]  22 00 FF FF FF FF 00 D8
  can1  19F0C401   [8]  23 72 46 FF FF FF FF FF
  can1  19F0C401   [8]  24 FF FF FF FF FF FF FF
  can1  19F0C401   [8]  25 00 FF FF FF FF FF FF

after it hits 1110 it goes back to 0000

tidbits of information

Susurrus, 2015-05-14:
Xanbus does use Fast Packets, as I decode PGN127173, which was transmit by the Xantrex XW-MPPT60-150 solar charger.

I have already done all of this work in that code I linked to above, you should read through it in case you haven't already. The fast packets are special in that the first two bytes of the first packet have special meaning, with the first byte having two parts: a sequence ID (the 3 most significant bits) and a frame counter (the 5 least-significant bits). The second byte is the total data length of the message (hence why the byte limit is 255). For messages after the first, only the first byte has special meaning, which is the sequence ID and frame counter just like in the first message's first byte.