xela144/CANaconda #87: I get no output from my radar at all

koenlek, 2015-05-11:
Hi all, 

I have the CANUSB +# v997 adapter by Lawicel and I am trying to read the output of an ARS 309-2 radar. Using CANaconda, I run ```./CANaconda.py -p /dev/ttyUSB0 --canbaud 500k -m metadata/koentest.xml```

Where koentest has a proper message definition in it, which all seems to work fine. Unfortunately, CANaconda doesn't show any output at all, even no raw output. 

Is there any way I can verify that my adapter / setup is working at all?
```cat -v < /dev/ttyUSB0``` doesn't output anything at all. Whereas it is supposed to output a bunch of crap at least I belief? 

I did not install the Lawicel drivers, as I don't think ubuntu 14.04 supports the 
```none /proc/bus/usb   usbfs   defaults,devmode=0666   0 0```
line that you should put in /etc/fstab. Can anyone point me in the right direction? 

Using ```sudo lsmod | grep ftdi``` shows that ```ftdi_sio``` and ```usbserial``` are loaded. They unload if I disconnect. I didn't install any special drivers. And I am sure that /dev/ttyUSB0 is the right device and that I am using the right baudrate.

Any help would be very much appreciated.

koenlek, 2015-05-11:
I noticed that the ARS 309-2 uses CANopen, might that be the issue?

koenlek, 2015-05-11:
Found it! I did not yet install can-utils yet! Now everything is working!

closed 2015-05-11 by koenlek

Susurrus, 2015-05-11:
If you have your `koentest.xml` defining enough messages for the radar to be usable, go ahead and submit a PR for it. If it's well-documented, we'd be happy to include it in the mainline metadata folder. Should make it easier on you once you have those messages all defined.